carla-simulator/scenario_runner #1021: No more scenarios .... Exiting

ravivaghasiya1998, 2023-08-31:
**
Hello 
I am getting the error when I try to run scenario
**

Did anyone have solution to this error?

I have set the environment variable as:

PYTHONPATH = D:\CARLA_0.9.13\WindowsNoEditor\PythonAPI\carla
CARLA_ROOT = D:\CARLA_0.9.13\WindowsNoEditor
SCENARIO_RUNNER_ROOT =D:\CARLA_0.9.13\scenario_runner

**[image]**

- OS: [Windows 10]
 - CARLA 0.9.13
 - Python 3.7
 - Version 0.9.13

ronyshaji, 2023-09-01:
Hey
Since you are running OpenScenario, you need to use the argument --openscenario

`Python scenario_runner.py --openscenario srunner/examples/FollowLeadingVehicle.xosc --output`

ravivaghasiya1998, 2023-09-01:
Thanks. It worked

ronyshaji, 2023-11-07:
@ravivaghasiya1998 Please close this issue