carla-simulator/scenario_runner #1027: OpenScenario 2.0 Path Modifiers

jasmeetgaba24, 2023-10-06:
**Trying to Use Path Modifiers in OSC File to run Scenario in Carla Simulator**

Hi,
I am trying to utilize Path Modifiers such as path_min_driving_lane or path_over_junction in my OSC Files, However Those modifiers are not making any change, It look like for every Carla Town - Location is Predefined, and No Matter which Lane, Junction, or Sign I used. vehicle is getting spawned at same location for every Town.

Is that implementation for Path Modifiers are not in place or Am I making any mistake.
Below is an OSC file I am trying to use.

CHANGE_LANE_SCENARIO:
It does not matter how many minimum lanes I keep, or Whatever Sign I used, Ego Vehicle gets spawned at same point.
I believe SET_MAP  is the only Path modifer which is working and loading the mentioned Town in carla.
"""
import basic.osc

scenario top:
    path: Path                      # A path in the map
    path.set_map("Town02")    # specify map to use in this test
    **path.path_min_driving_lanes(2)         # Path should have at least two lanes
    path.path_has_sign("stop_sign")**

    ego_vehicle: Model3                # ego car
    npc: Rubicon               # The other car

    event start
    event end
    event left
    event right
    do serial:
        get_ahead: parallel(duration: 15s):
            ego_vehicle.drive(path) with:
                speed(20kph)
                lane(1, at: start) # left to right: [1..n]
            npc.drive(path) with:
                lane(right_of: ego_vehicle, at: start)
                position(15m, behind: ego_vehicle, at: start)
                position(20m, ahead_of: ego_vehicle, at: end)

        change_lane: parallel(duration: 5s):
            ego_vehicle.drive(path)
            npc.drive(path) with:
                # change_lane(lane_changes:[1..2], side:left)
                change_lane(lane_changes:[1..2], side: left)

        slow: parallel(duration: 20s):
            ego_vehicle.drive(path)
            npc.drive(path) with:
                speed(20kph)

"""

jasmeetgaba24, 2023-10-06:
I've tried debugging and it can bee seen, Path has the values for _min_driving_lanes and _signs_type, However its not reflecting while running the senario.

[image]

jasmeetgaba24, 2023-10-09:
@charles-lei, @glopezdiest Can you provide some insights on same ?

kshitiz-deloitte, 2023-10-11:
I had the same doubt how it was taking the path modifiers as @jasmeetgaba24 mentioned.

whdtjr, 2024-07-31:
I have similar challenge
did you get the solution??

4l3x4ndre, 2024-09-20:
@jasmeetgaba24 do you have any news ?