carla-simulator/scenario_runner #1030: ScenarioRunner environment variables and Python paths for Carla in Docker Container

jakobfrosch, 2023-11-14:
**Describe our question or idea**
I want to run Carla Scenario Runner with an Apollo Carla bridge in which it is necessary that Carla runs in a Docker container. Now I still want to run Scenario Runner and am (probably) failing to set the environment variables.
According to the instructions, the variables should be initialized as follows:
```
export CARLA_ROOT=/path/to/your/carla/installation
export SCENARIO_RUNNER_ROOT=/path/to/your/scenario/runner/installation
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-<VERSION>.egg
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
```

The question, however, is what is the /path/to/your/carla/installation when it runs in the Docker container?
In the running Carla container it is /home/carla.
**Expected behavior**
Scenario runner runs after executing python3.7 scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

Current error: 
AttributeError: module 'tensorflow' has no attribute 'contrib'

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.2 LTS
 - CARLA Version: CARLA 0.9.14
 - Python version: 3.7
 - Version: 0.9.13

romanmatus, 2024-03-05:
Hi, did you figure out ?

whatgit, 2024-03-25:
Same problem here

lei3li, 2024-04-17:
same problem here, did anyone find the solution?

Enginerd0, 2024-07-04:
i also have this issue

MDCmarc, 2025-02-07:
For future reference, altough not the most efficient, you can just cp the pythonAPI of carla into the scenario_runner folder.
`docker cp container:carla/PythonAPI/carla scenario_runner/`
And then just add
`echo 'export PYTHONPATH=$PYTHONPATH:/apollo/scenario_runner/carla' >> ~/.bashrc`
`echo 'export SCENARIO_RUNNER_ROOT=/apollo/scenario_runner' >> ~/.bashrc`