carla-simulator/scenario_runner #1047: Running OpenScenario files from RoadRunner in ScenarioRunner

haabe, 2024-01-12:
I know RoadRunner is used by the Carla team, as the newest Town OpenDrive files originates from RoadRunner. So I assumed that running OpenScenario files exported from RoadRunner in Carl/ScenarioRunner would work.

My initial attempt was to import a townX.xodr from Carla into RoadRunner, generate maps and then a scenario. Then export the scenario to OpenScenario and execute it in ScenarioRunner. This did not work as expected. The car spawned at a totally different location and drove off in the wrong direction.

I checked a little more, and found that even though I checked the lat + lon values of the original OpenDrive file matched the lat + lon set up in RR, RR did not match the north, south, east, west values in the header. So I assumed that even though the world location matched, the imported map was misaligned.

Upon investigating further, I created some simple maps in RR (a 3 x 3 grid of roads and intersections and a big circular road), and then creating simple scenarios for those maps. When executing these scenarios in Carla/ScenarioRunner, I experienced the same errors - the cars spawned in the wrong location and drove erratically.

I first tried RoadRunner support at Mathworks, and they assume that Carla/ScenarioRunner doesn't comply with the ISO8855:2011 standards that OpenScenario should use, or that it interprets it incorrectly.

Have anyone successfully managed to use OpenScenario files from RR in ScenarioRunner, or have any other ideas an suggestions on how to make this work?

starrye, 2024-02-18:
Perhaps you can try looking at the xodr information at https://odrviewer.io/ and determine the coordinates

cdcstar, 2025-03-24:
i have try to run ScenarioRunner use OpenScenario files from roadrunner , it can not support all, some actions and trigger can not run ,

cdcstar, 2025-03-24:
@haabe  do you have scuessful?