carla-simulator/scenario_runner #1054: Near-Miss Test Criteria

paulbourelly999, 2024-02-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
My team is attempting to improve our simulation environment by taking advantage of CARLA Scenario Runner to evaluate run scenarios for collisions and traffic violations. Additionally we are testing vehicle perception and cooperative perception performance for improving safety and mobility. For this we would like a near-miss criteria similar to the collision test. 

Reading the documentation it seemed that we should create our own metric similar to distance between two actors but this is none ideal since it only records some quantitative value like distance instead of returning information about when a given distance threshold is violated.

Ideally we would want somethings similar to collision test but for pedestrian near-misses where we get a report after the scenario that will tell use whether we ever came two close to a pedestrian on the road and return information about these instances.

I assume the best way to create this maybe to write my own Criteria that constantly checks actor distance to pedestrians and if that ever gets below a threshold and the pedestrian is not on the sidewalk (not sure if this information is available) log some sort of information about the event. 

Is this the correct way to proceed? Any help would be greatly appreciated.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

starrye, 2024-02-18:
Do you want to make a feature that is evaluated after simulation? These are not included in the simulation process, you may use `start_recorder` to perform a data record, after simulation by analyzing the log to achieve

paulbourelly999, 2024-02-19:
Ideally I would want something like CollisionTest criteria that is evaluated without replaying log.  If I wrote a new class to extend Criterion that just checked the distance between a given actor and any pedestrian and ensured that it never fell below a given safe threshold.