carla-simulator/scenario_runner #1061: Raise KeyError 'misc' when using MiscObject in OpenScenario files

PhyllisJi, 2024-02-29:
**Describe the bug**
The OpenScenario file containing the MiscObject keyword could not complete compilation

**Expected behavior**
Compiles properly

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10]
 - CARLA Version [e.g. CARLA 0.9.13]
 - Python version [e.g. 3.10]
 - Version [e.g. 0.9.13]

**Additional context**
Add any other context about the problem here.

starrye, 2024-03-11:
Can I see your xosc file?

PhyllisJi, 2024-03-11:
> Can I see your xosc file?

Of Course.

<?xml version="1.0" encoding="utf-8"?>
<OpenScenario>
  <FileHeader revMajor="1" revMinor="0" date="2024-03-11T16:19:49" description="CARLA:test" author=""/>
  <CatalogLocations>
  </CatalogLocations>
  <RoadNetwork>
    <LogicFile filepath="Town05"/>
  </RoadNetwork>
  <Entities>
    <ScenarioObject name="hero">
      <Vehicle name="vehicle.lincoln.mkz_2017" vehicleCategory="car">
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties/>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="adversary">
      <Vehicle name="vehicle.tesla.model3" vehicleCategory="car">
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties/>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="Box">
      <MiscObject name="box" miscObjectCategory="obstacle" model3d="../models/box_cc_by.osgb" mass="200.0">
        <BoundingBox>
          <Center x="0.0" y="0.0" z="0.75"/>
          <Dimensions width="1.0" length="2.0" height="1.2"/>
        </BoundingBox>
        <Properties/>
      </MiscObject>
    </ScenarioObject>
  </Entities>
  <Storyboard>
    <Init>
      <Actions>
        <GlobalAction>
          <EnvironmentAction>
            <Environment name="weather">
              <RoadCondition frictionScaleFactor="1.0"/>
              <Weather>
                <Precipitation precipitationType="dry"/>
                <Fog visualRange="100000.0"/>
                <Sun intensity="0.85" elevation="0.5" azimuth="0.5"/>
              </Weather>
              <TimeOfDay dateTime="2020-01-01T12:00:00" animation="true"/>
            </Environment>
          </EnvironmentAction>
        </GlobalAction>
        <Private entityRef="hero">
          <PrivateAction>
            <ControllerAction>
[... 128 lines cut ...]
                </ConditionGroup>
              </StartTrigger>
            </Event>
          </Maneuver>
        </ManeuverGroup>
        <StartTrigger/>
        <StopTrigger>
          <ConditionGroup>
            <Condition name="EndCondition" delay="0" conditionEdge="rising">
              <ByEntityCondition>
                <TriggeringEntities triggeringEntitiesRule="any">
                  <EntityRef entityRef="hero"/>
                </TriggeringEntities>
                <EntityCondition>
                  <TraveledDistanceCondition value="200.0"/>
                </EntityCondition>
              </ByEntityCondition>
            </Condition>
          </ConditionGroup>
        </StopTrigger>
      </Act>
    </Story>
    <StopTrigger/>
  </Storyboard>
</OpenScenario>

PhyllisJi, 2024-03-11:
> Can I see your xosc file?
```xml
<?xml version="1.0" encoding="utf-8"?>
<OpenScenario>
  <FileHeader revMajor="1" revMinor="0" date="2024-03-11T16:19:49" description="CARLA:test" author=""/>
  <CatalogLocations>
  </CatalogLocations>
  <RoadNetwork>
    <LogicFile filepath="Town05"/>
  </RoadNetwork>
  <Entities>
    <ScenarioObject name="hero">
      <Vehicle name="vehicle.lincoln.mkz_2017" vehicleCategory="car">
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties/>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="adversary">
      <Vehicle name="vehicle.tesla.model3" vehicleCategory="car">
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties/>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="Box">
      <MiscObject name="box" miscObjectCategory="obstacle" model3d="../models/box_cc_by.osgb" mass="200.0">
        <BoundingBox>
          <Center x="0.0" y="0.0" z="0.75"/>
          <Dimensions width="1.0" length="2.0" height="1.2"/>
        </BoundingBox>
        <Properties/>
      </MiscObject>
    </ScenarioObject>
  </Entities>
  <Storyboard>
    <Init>
      <Actions>
        <GlobalAction>
          <EnvironmentAction>
            <Environment name="weather">
              <RoadCondition frictionScaleFactor="1.0"/>
              <Weather>
                <Precipitation precipitationType="dry"/>
                <Fog visualRange="100000.0"/>
                <Sun intensity="0.85" elevation="0.5" azimuth="0.5"/>
              </Weather>
              <TimeOfDay dateTime="2020-01-01T12:00:00" animation="true"/>
            </Environment>
          </EnvironmentAction>
        </GlobalAction>
        <Private entityRef="hero">
          <PrivateAction>
            <ControllerAction>
              <AssignControllerAction>
                <Controller name="HeroAgent">
[... 127 lines cut ...]
              </StartTrigger>
            </Event>
          </Maneuver>
        </ManeuverGroup>
        <StartTrigger/>
        <StopTrigger>
          <ConditionGroup>
            <Condition name="EndCondition" delay="0" conditionEdge="rising">
              <ByEntityCondition>
                <TriggeringEntities triggeringEntitiesRule="any">
                  <EntityRef entityRef="hero"/>
                </TriggeringEntities>
                <EntityCondition>
                  <TraveledDistanceCondition value="200.0"/>
                </EntityCondition>
              </ByEntityCondition>
            </Condition>
          </ConditionGroup>
        </StopTrigger>
      </Act>
    </Story>
    <StopTrigger/>
  </Storyboard>
</OpenScenario>
```

starrye, 2024-03-11:
You can test the `box` by Carla whether there is or you can replace a default static object:`static.prop.streetbarrier`

PhyllisJi, 2024-04-12:
> you can change your xosc file to test whether the `box` exists
> 
> ```
> <ScenarioObject name="Box">
>       <MiscObject name="box" miscObjectCategory="obstacle" model3d="../models/box_cc_by.osgb" mass="200.0">
>         <BoundingBox>
>           <Center x="0.0" y="0.0" z="0.75"/>
>           <Dimensions width="1.0" length="2.0" height="1.2"/>
>         </BoundingBox>
>         <Properties/>
>       </MiscObject>
>     </ScenarioObject>
> ```
> 
> ===>
> 
> ```
> <ScenarioObject name="Box">
>       <MiscObject name="static.prop.streetbarrier" miscObjectCategory="obstacle" mass="0.0">
>         <BoundingBox>
>           <Center x="0.0" y="0.0" z="0.75"/>
>           <Dimensions width="1.0" length="2.0" height="1.2"/>
>         </BoundingBox>
>         <Properties/>
>       </MiscObject>
>     </ScenarioObject>
> ```

It works！Thank you!

closed 2024-04-12 by PhyllisJi