carla-simulator/scenario_runner #1069: The stop trigger on the storyboard is not working.

hlsongg, 2024-04-05:
I made an example of the situation below. 

[Example]
1. The Ego vehicle starts at SimulationTime = 0 and increases the speed from 0 to 60 km/h. (Accelerate for 5 seconds)
2. In front of the lane where the Ego vehicle is located, there is a counter vehicle.
3. If the distance between the Ego vehicle and the red_Vehicle is narrowed to 30 m, the speed of the Ego vehicle is reduced to 30 km/h.
4. Continue to drive both vehicles for 1 minute before terminating.

I have a OpenSCENARIO file which has scenario end condition (StopTrigger on StoryBoard) as StoryboardElementStateCondition converting on endTransition. However, after SlowPrecordingStartCondition is triggered, the stop trigger does not work and ends with the following error. Why?
`Destroying ego vehicle 147`
`ERROR: failed to destroy actor 147 : unable to destroy actor: not found `

`<?xml version="1.0"?>

<OpenSCENARIO>
  <FileHeader revMajor="1" revMinor="0" date="2024-03-25T12:00:00" description="CARLA:test_0322" author=""/>
  <ParameterDeclarations/>
  <CatalogLocations/>
  <RoadNetwork>
    <LogicFile filepath="Town04"/>
    <SceneGraphFile filepath=""/>
  </RoadNetwork>
  <Entities>
    <ScenarioObject name="hero">
      <Vehicle name="vehicle.tesla.model3" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="120" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="ego_vehicle"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
    <ScenarioObject name="standing">
      <Vehicle name="vehicle.volkswagen.t2" vehicleCategory="car">
        <ParameterDeclarations/>
        <Performance maxSpeed="69.444" maxAcceleration="200" maxDeceleration="10.0"/>
        <BoundingBox>
          <Center x="1.5" y="0.0" z="0.9"/>
          <Dimensions width="2.1" length="4.5" height="1.8"/>
        </BoundingBox>
        <Axles>
          <FrontAxle maxSteering="0.5" wheelDiameter="0.6" trackWidth="1.8" positionX="3.1" positionZ="0.3"/>
          <RearAxle maxSteering="0.0" wheelDiameter="0.6" trackWidth="1.8" positionX="0.0" positionZ="0.3"/>
        </Axles>
        <Properties>
          <Property name="type" value="simulation"/>
        </Properties>
      </Vehicle>
    </ScenarioObject>
  </Entities>

  <Storyboard>

    <Init>
      <Actions>	  

        <GlobalAction> 
          <EnvironmentAction>
            <Environment name="Environment1">
              <TimeOfDay animation="false" dateTime="2024-03-25T12:00:00"/>
              <Weather cloudState="free">
                <Sun intensity="0.85" azimuth="0" elevation="1.31"/>
                <Fog visualRange="100000.0"/>
                <Precipitation precipitationType="dry" intensity="0.0"/>
              </Weather>
              <RoadCondition frictionScaleFactor="4.0"/>
            </Environment>
          </EnvironmentAction>
        </GlobalAction>

        
        <Private entityRef="hero">
          <PrivateAction> 
            <TeleportAction>
              <Position>
                <WorldPosition x="-9.4" y="-152.8" z="0.5" h="1.57079632679"/>
              </Position>
            </TeleportAction>
          </PrivateAction>

          <PrivateAction>
            <LongitudinalAction>
              <SpeedAction> 
                <SpeedActionDynamics dynamicsShape="linear" dynamicsDimension="rate" value="4.0"/> 
                  <SpeedActionTarget> 
                    <AbsoluteTargetSpeed value="16.6667"/> 
                  </SpeedActionTarget>
              </SpeedAction>
            </LongitudinalAction>
          </PrivateAction>

        </Private>

			
        <Private entityRef="standing">
          <PrivateAction>
            <TeleportAction> 
              <Position>
                <WorldPosition x="-9.4" y="-2.8" z="0.5" h="1.57079632679"/>
              </Position>
            </TeleportAction>
          </PrivateAction>

          <PrivateAction>
            <LongitudinalAction>
              <SpeedAction>
                <SpeedActionDynamics dynamicsShape="step" dynamicsDimension="time" value="0.0"/>
                <SpeedActionTarget>
                  <AbsoluteTargetSpeed value="8.33333"/> 
                </SpeedActionTarget>
              </SpeedAction>
            </LongitudinalAction>
          </PrivateAction>
        </Private>
      </Actions>
    </Init>

    <Story name="SlowPrecedingStory"> 
      <Act name="SlowPrecedingAct">
        	      <ManeuverGroup maximumExecutionCount="1" name="SlowPrecedingSequence">
          <Actors selectTriggeringEntities="false">
            <EntityRef entityRef="hero"/>
          </Actors>
          <Maneuver name="SlowPrecedingManeuver">
            <Event name="SlowPrecedingEvent" priority="overwrite">
              <Action name="SlowPrecedingAction"> 
                <PrivateAction>
                
                  <LongitudinalAction>
                    <SpeedAction>
                      <SpeedActionDynamics  dynamicsShape="linear" 
                                            dynamicsDimension="rate"
                                            value="16.0"/> 
                      <SpeedActionTarget>
                        <AbsoluteTargetSpeed value="8.33333"/> 
                      </SpeedActionTarget>
                    </SpeedAction>
                  </LongitudinalAction>
                </PrivateAction>
              </Action>
              
              
              <StartTrigger> 
                <ConditionGroup>
                
                  <Condition name="SlowPrecedingStartCondition" delay="0" conditionEdge="rising">
					          <ByEntityCondition>
                      <TriggeringEntities triggeringEntitiesRule="any">
                        <EntityRef entityRef="hero"/>
                      </TriggeringEntities>
                      <EntityCondition> 
									    
									      <RelativeDistanceCondition 	entityRef="standing"
												                    				relativeDistanceType="longitudinal"
															                    	value="30"
													                    			freespace="true"
														                    		rule="lessThan"/>
								      </EntityCondition>
                    </ByEntityCondition>
                  </Condition>
                </ConditionGroup>
              </StartTrigger>
            </Event>
          </Maneuver>
        </ManeuverGroup>
			
			
        <StartTrigger> 
          <ConditionGroup>
            <Condition name="SlowPrecedingActStart" delay="0" conditionEdge="none">
              <ByValueCondition>
                <SimulationTimeCondition value="0" rule="greaterThan"/> 
              </ByValueCondition>
            </Condition>
          </ConditionGroup>
        </StartTrigger>
      </Act>
    </Story>
    
    <StopTrigger> 
      <ConditionGroup>
        
        <Condition name="ActStopCondition" delay="60" conditionEdge="rising">
         
          <ByValueCondition> 
            <StoryboardElementStateCondition storyboardElementType="maneuver" storyboardElementRef="SlowPrecedingManeuver" state="endTransition"/>
          </ByValueCondition>
        </Condition>
      </ConditionGroup>
    </StopTrigger>
  </Storyboard>
</OpenSCENARIO>`

dx6nxr, 2024-10-11:
If I remember correctly, endTransition is not supported as for now.

Best, Ivan

dx6nxr, 2024-10-16:
> If I remember correctly, endTransition is not supported as for now.
> 
> Best, Ivan

Following on my previous reply: endTransition is indeed supported, but i not treated as "tansition" so to say. In openscenario_parser.py, 
[image]
All of them are treated equally, namely: the finish of the manueverGroup in this case. 

Considering the trigger problem:

In your case it is following:
Vehicles move, when the distance changes to <30 meters, the manueverGroup is being executed (the one inside of SlowPrecedingStory). As it is the only manueverGroup present in the file, the simulation just ends when the manueverGroup is fully executed, as there are no other manueverGroups to execute and therefore stopTrigger is being "ignored" as there's nothing more to run and therefore to stop.

Solution: add another manueverGroup that keeps vehicle driving for another minute.

Hope this helps.
Best,
Ivan