carla-simulator/scenario_runner #1084: OpenSCENARIO 2.0 questions

yarinr25, 2024-05-29:
hey everyone I have some questions about using OpenSCENARIO 2.0 
1) Is it not possible to use the standard.osc file from the Asam documentation?
2) I saw that there are limitations when writing osc scenarios, there is documentation that summarizes what can be written and what not? 
3) Is it possible to use pedestrians in osc? I have tried to add it as an actor to the basic.osc file but it is not recognized.

fangchaooo, 2024-06-04:
I have same question.
How can a person without background knowledge in this area check the correctness of an ANTLR4 .g4 file?

There are three versions for osc2 definition file:

1. https://github.com/PMSFIT/py-osc2/blob/main/osc2parser/openscenario2.g4
2. https://github.com/51WORLD/osc2checker/blob/main/osc2grammar/OpenSCENARIO2.g4
3. this repo
I don't know which is completely correct

yarinr25, 2024-06-04:
@fangchaooo if your goal is to check syntax and correctness of .osc files, I recommend using VS code and installing the OpenSCENARIO extensions.

fangchaooo, 2024-06-04:
@yarinr25 My goal is to check parser and lexer correctly for osc2 language.