carla-simulator/scenario_runner #1088: OSC2ScenarioConfiguration' object has no attribute 'route'

zwhaa, 2024-06-20:
When I run the python scenario_runner.py --openscenario2 srunner/examples/xxx.osc command, no matter which file I run, I will report an error：OSC2ScenarioConfiguration' object has no attribute 'route' 'OSC2ScenarioConfiguration' object has no attribute 'route' No more scenarios .... Exiting

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - CARLA Version [e.g. CARLA 0.9.15]
 - Python version [e.g. 3.7]
 - Version [e.g. 0.9.15]

glopezdiest, 2024-06-21:
Hey @zwhaa please check if you have [this line](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenarioconfigs/openscenario_configuration.py#L37) in your code, as it is the one that fixes the issue. It was part of the 0.9.15 release though, so not sure why you wouldn't have it.

Feel free to reopen the issue if it is still happening

closed 2024-06-21 by glopezdiest

zwhaa, 2024-06-24:
Hey @glopezdiest This line of code is included in the code. I tried to add the following content to the osc2scenariounconfiguration. py file according to the error prompt, and it ran without any problems.