carla-simulator/scenario_runner #1092: ASAM standard.osc support in osc2 scenario_runner

dx6nxr, 2024-06-27:
Hi, following on my previous issue, namely: [here](https://github.com/carla-simulator/scenario_runner/issues/1090#issue-2374650915), and also issue made by another community member [here](https://github.com/carla-simulator/scenario_runner/issues/1084#issue-2322636674).
I've been trying to make Scenario runner to work with standard.osc library provided by ASAM, so far I've had tough luck doing it, as it will spit a bunch of syntaxis errors in the standard.osc itself, and will not run a scenario. Current state can be seen on this screenshot: 
[image]
which is pretty odd to me, as officially distributed library must not contain syntax errors. Was scenario_runner even tested with standard.osc? As there are no example or test files using this library, and the instruments that basic.osc provides are pretty "basic", and are not enough for my use case. Is there a way to use standard.osc or will I have to modify basic.osc in a way I need it to be? 
Thanks in advance.

Best regards,
Ivan Burmistrov

Desktop:
 - OS: win11
 - CARLA: 0.9.15
 - Python version: 3.8.10
 - Version: latest

DimK19, 2024-08-30:
I have the same problem. For reference, the file `standard.osc` is provided by ASAM at this webpage, under “Additional resources”: https://www.asam.net/static_downloads/public/asam-openscenario/2.0.0/domain-model/standard_library.html

@dx6nxr Have you found any solution? Thanks.

dx6nxr, 2024-09-01:
Hey @DimK19,
no, as far as I understood after looking at the files there's no way to make it work without major changes to scenario runner core. Additionally, scenario_runner support for osc2 does not support any triggers (i.e. distance triggers etc), which are very needed for my use-case, so I just settled down on using osc1.0 with all of it's underlying restrictions, but at least it works xD

DimK19, 2024-09-02:
> there's no way to make it work without major changes to scenario runner core.

I have reached the same conclusion. As yet, osc2 is not sufficiently supported.

hsiang-lee, 2026-01-05:
> Have we confirmed that `import` statements in osc2 files are supported yet in scenario runner? That is, if I provide the file `osc.standard` in the right location, and an `.osc` file tries to `import osc.standard`, will the definitions get read in by scenario runner?

[https://osc2studio.online](https://osc2studio.online)