carla-simulator/scenario_runner #1097: ModuleNotFoundError: No module named 'agents.navigation'

Enginerd0, 2024-07-08:
**Describe our question or idea**
Hello,

When I try to run scenario_runner.py using the Getting Started documentation example I get an error that there is no module named agents.navigation found. Any ideas why I would be getting this error?

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Command line input:
# Inside the ScenarioRunner root directory
python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Error:
Traceback (most recent call last):
  File "scenario_runner.py", line 41, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "C:\Users\Desktop\CARLA_0.9.15\WindowsNoEditor\PythonAPI\examples\srunner\scenarioconfigs\openscenario_configuration.py", line 24, in <module>
    from srunner.scenariomanager.carla_data_provider import CarlaDataProvider  # workaround
  File "C:\Users\Desktop\CARLA_0.9.15\WindowsNoEditor\PythonAPI\examples\srunner\scenariomanager\carla_data_provider.py", line 22, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents.navigation'

 - OS: Windows 10
 - CARLA Version 0.9.15
 - Python version 3.7
 - Version 0.9.15

Enginerd0, 2024-07-10:
Fixed the issue

closed 2024-07-10 by Enginerd0

BibiBea, 2024-07-11:
Can you please share how you fixed it?

starrye, 2024-11-14:
> Can you please share how you fixed it?

https://github.com/carla-simulator/carla/tree/e41ffa03c27e381f52bc98402cc3100aac225ac2/PythonAPI/carla/agents