carla-simulator/scenario_runner #1103: where is the 0.9.14version?

chan-yuu, 2024-07-26:
**Describe our question or idea**
A clear and concise description.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: ubuntu20
 - CARLA 0.9.14
 - Python version3.7
 - Version 0.9.14

Daraan, 2024-09-09:
I cannot answer why there is no 0.9.14 version.

For me the 0.9.15 version also works with 0.9.14, but I am using not all aspects of it so I cannot tell if everything will work without hick-ups.

If you face problems going back to this PR https://github.com/carla-simulator/scenario_runner/pull/1082 might help

PabloVD, 2024-12-02:
We have explicitly stated it in the readme for further clarification.