carla-simulator/scenario_runner #1107: cant run open scenario

Turuu1124, 2024-08-16:
OS: WIndows 11
Carla: 0.9.15-packaged version
Python: 3.7 - anaconda 
Hello,
I am having trouble with running the open scenario as i keep getting the No module named 'agents', I tried to fix this by using this post: [896](https://github.com/carla-simulator/scenario_runner/issues/896) as shown in the image below, but even after that i still doesn't work i read through the [getting started with scenario_runner ](https://scenario-runner.readthedocs.io/en/latest/getting_scenariorunner/) and did the packaged version steps fully and followed the steps but i still cant run it. Any help would be appreciated thank you.
[image]

starrye, 2024-11-14:
https://github.com/carla-simulator/carla/tree/e41ffa03c27e381f52bc98402cc3100aac225ac2/PythonAPI/carla/agents

minhaj6, 2025-01-02:
just copy the agents folder from carla simulator and paste it in the root of the scenario runner folder.