carla-simulator/scenario_runner #1109: What software is used to write OpenScenario2.0 scenes？

Xiaowuzjj, 2024-08-27:
What software is used to write OpenScenario2.0 scenes, and what are the scene language specifications of version 2.0?

mljack, 2024-10-13:
Specs:
https://publications.pages.asam.net/standards/ASAM_OpenSCENARIO/ASAM_OpenSCENARIO_DSL/latest/introduction.html

JesmonX, 2025-03-24:
> Specs (which is 2.1, while scenario_runner supports 2.0): https://publications.pages.asam.net/standards/ASAM_OpenSCENARIO/ASAM_OpenSCENARIO_DSL/latest/introduction.html

I can't find any information of Specs, could you please provide me an url  about it?

mljack, 2025-03-26:
[image]

mljack, 2025-03-26:
[image]

mljack, 2025-03-26:
For "specs", I mean all details of OpenSCENARIO DSL in the url I mentioned above.

hsiang-lee, 2025-12-04:
OpenSCENARIO 2.0 is the latest scenario description standard and represents the future direction of the field. It utilizes a Domain-Specific Language (DSL) for authoring scenarios. There is an online tool available that supports writing in this standard, offering features such as syntax highlighting, code auto-completion, and real-time error checking. Currently, Scenario Runner provides support for a subset of the features defined in OpenSCENARIO 2.0

[https://www.osc2studio.online](https://www.osc2studio.online)