carla-simulator/scenario_runner #1117: Do we have a list of feature coverage on OpenSCENARIO 1.0 AND 2.0

mljack, 2024-10-11:
I have seen a few issues mentioned for OpenSCENARIO features and a milestone for 1.0. (https://github.com/carla-simulator/scenario_runner/issues/1090 https://github.com/carla-simulator/scenario_runner/issues/1092 https://github.com/carla-simulator/scenario_runner/issues/1084 https://github.com/carla-simulator/scenario_runner/milestone/3)

Do we have a list of feature coverage on OpenSCENARIO 1.0 AND 2.0, something like the esmini project provided?
https://github.com/esmini/esmini/blob/master/osc_coverage.txt

It would be quite helpful to clarify supported functionalities and room for improvements.

dx6nxr, 2024-10-21:
List of supported and unsupported features for osc1.0 can be found in the file "openscenario_support.md" ([link](https://github.com/carla-simulator/scenario_runner/blob/master/Docs/openscenario_support.md)). AFAIK there is no such file for osc2.0.

hsiang-lee, 2026-01-05:
[https://osc2studio.online](https://osc2studio.online)