carla-simulator/scenario_runner #1120: Issues with using scenario_runner.py for versions 0.9.15

NayaBaslan, 2024-10-16:
**Version 0.9.15 for scenario_runner not working**
I have CARLA 0.9.15 running and I need to use scenario_runner. According to the instructions in this repo, I need to install the corresponding version of scenario_runner: Version 0.9.15
To run CARLA:
`./CarlaUE4.sh` 
and the scenario_runner:
`python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld`

However `scenario_runner.py` is not working and giving the errors:
` from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents' `

**Desktop**
 - OS: [Ubuntu 20.04]
 - CARLA Version [ CARLA 0.9.15]
 - Python version [3.8.10 ]
 - Version [0.9.15]

Can someone please help me on this?
As if the path to carla is not regognized, but the instructions for this are only for building from source.
It should directly work when downloading from packages.

NayaBaslan, 2024-10-16:
however, when I try the `python3 no_rendering_mode.py` from scenario_runner, it works well and it connected to CARLA.
So the issue only seems for the agents. 

On the other hand, when I use a basic or behavior agent directly in the CARLA directory, for example in the `automatic_contol.py` script in CARLA, it works without problems.

I face the issue only when running ` python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld `

NayaBaslan, 2024-10-17:
solved by copying agents directory into scenario_runner directory.

closed 2024-10-17 by NayaBaslan

minhaj6, 2025-01-01:
Simplest solution. This should be in the docs. 👍🏻

HuangChiEn, 2025-08-07:
wow, v0.9.15, the latest released version, even have this kind of mistake ?!
placing dir to the wrong place ?!