carla-simulator/scenario_runner #1123: ScenarioRunner crashes when trying to spawn a new car at runtime.

vollkornholztuer, 2024-10-18:
**Describe our question or idea**
ScenarioRunner crashes when excecuting an OpenSCENARIO-1.0 file, which tries to spawn a new car while the simulation is already running. The Cars that spawn in the Init-section, and therefore are present from the beginning, work. It kind of works like this:

I run the .xosc with ScenarioRunner, the simulations starts with two cars (Ego and Target). The target car begins to drive as predefined, the Ego car is stationary. As soon as the new car is supposed to spawn, the new car is visible for a single frame but it crashes. The error message from the terminal sadly is not very descriptive (see screenshots).

**Expected behavior**
The car should spawn and the simulation should keep running and finish at some point.

**Screenshots and Screencasts**
The error message from the terminal, when the scenario is executed:
[image]

A video of the execution of the given OpenSCENARIO-File (see "Other resources"):
[Screencast from 10-18-2024 12:44:33 PM.webm](https://github.com/user-attachments/assets/6a2beb09-9f30-4d5a-ae68-65c4aad58916)

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS
 - CARLA Version: CARLA 0.9.15
 - Python version: 3.10
 - Version: 0.9.15
 
 - CPU: Intel i9-12900
 - GPU: Intel UHD Graphics 770 (iGPU)
 - RAM: 64GB

 I read in another issue (https://github.com/carla-simulator/scenario_runner/issues/793), that not having an Nvidia GPU might lead to crashes, although that specific issue doesn't happen to me. That's why i included the most important hardware.

**Other resources**
My OpenSCENARIO 1.0-file: [spawnDespawnTest0_copy.zip](https://github.com/user-attachments/files/17434264/spawnDespawnTest0_copy.zip)

vollkornholztuer, 2024-10-23:
I've found the "--debug" command for the cli. This is the ouptut, when running the provided .xosc file:
[spawnDespawnText0_debug.txt](https://github.com/user-attachments/files/17488008/spawnDespawnText0_debug.txt)

I'm trying to read it, but it's very cryptic. I probably won't understand it on my own.

starrye, 2024-11-04:
I don't quite understand your approach. I think the new car was actually generated at the beginning， but the location is not `<WorldPosition x="-8.0" y="0.0" z="0.0" h="0.0" p="0.0" r= "0.0" />`. Maybe you can print all the actor information after generation, or adjust the coordinates of new_car?

starrye, 2024-11-04:
I doubt there is a bug, you can update `srunner scenariomanager/scenarioatomics/atomic_behaviors`  `class AddActor`  、`update` method to verify.
```
    def update(self):
        new_status = py_trees.common.Status.RUNNING
        try:
            new_actor = CarlaDataProvider.request_new_actor(
                self._actor_type, self._spawn_point, color=self._color)
            if new_actor:
                new_status = py_trees.common.Status.SUCCESS
        except:  # pylint: disable=bare-except
            print("ActorSource unable to spawn actor")
            new_status = py_trees.common.Status.FAILURE
        finally:
            return new_status
```

vollkornholztuer, 2024-11-04:
I tried it out and it worked. The car now spawns, stays there and doesn't crash the simulation. Thank you very much :)

closed 2024-11-04 by vollkornholztuer