carla-simulator/scenario_runner #1128: Python package version problems

ClemensLinnhoff, 2024-11-13:
**Describe the bug**
I am trying to install the scenario runner under Ubuntu 22.04. I cloned the repository and wanted to install the requirements in a virtual environment.

However, I get an error from numpy:

```
ModuleNotFoundError: No module named 'distutils.msvccompiler'
```

I tried to get around that by using a newer numpy version: 1.20 instead of 1.18. Then, I did not get the error anymore.

But I get the following new errors:
```
ERROR: Could not find a version that satisfies the requirement opencv-python==4.2.0.32 (from versions: 3.4.0.14, 3.4.10.37, 3.4.11.39, 3.4.11.41, 3.4.11.43, 3.4.11.45, 3.4.13.47, 3.4.15.55, 3.4.16.57, 3.4.16.59, 3.4.17.61, 3.4.17.63, 3.4.18.65, 4.3.0.38, 4.4.0.40, 4.4.0.42, 4.4.0.44, 4.4.0.46, 4.5.1.48, 4.5.3.56, 4.5.4.58, 4.5.4.60, 4.5.5.62, 4.5.5.64, 4.6.0.66, 4.7.0.68, 4.7.0.72, 4.8.0.74, 4.8.0.76, 4.8.1.78, 4.9.0.80, 4.10.0.82, 4.10.0.84)
ERROR: No matching distribution found for opencv-python==4.2.0.32
```

So I tried to change the opencv version from 4.2.0.32 to 4.3.0.38, as this was the closest match. But then I get a bunch of other errors. 

When I remove the versions from numpy and opencv-python in the requirements, the requirements get installed. But then I get a Segmentation fault when trying to execute `python scenario_runner.py --help`.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup venv and source it
2. Run `pip3 install -r requirements.txt`

**Expected behavior**
I expected the requirements to be installed without any errors and the scenario_runner to run without a segmentation fault.

starrye, 2024-11-14:
https://github.com/carla-simulator/scenario_runner/pull/1121

PabloVD, 2024-11-29:
I couldn't reproduce the issue. I have created a conda environment with python 3.7 (the python version of the wheel included in the CARLA 0.9.15 package) and installed there the CARLA wheel and the scenario runner requirements of version 0.9.15. No errors were raised during the installation, and `python scenario_runner.py --help` was executed as expected without segmentation fault. So your mismatches between packages may come from using a different python version, please try with an 3.7 environment to check if that fixes it.